zxl19/Hand_Eye_Extrinsic_Calibration #2: Hi,Could you teach me how to calibrate the imu with the lidar

AdamPengG, 2021-11-03:


zxl19, 2021-11-04:
Hi @AdamPengG ! In our case we used an inertial navigation system (GNSS+IMU) to estimate 6 DoF pose. If you can estimate IMU pose and output in [this format](https://github.com/zxl19/Hand_Eye_Extrinsic_Calibration#2-input-pose-format), then you can config and run `main_calibration_L2I_quat_interp.m ` to see the results. You can refer to [this README file](https://github.com/zxl19/Hand_Eye_Extrinsic_Calibration/tree/main/tools) file to prepare your data.

AdamPengG, 2021-11-04:
Thanks Z. Could I put the ros imu message's quaternion to represent the INS' quaternion.

zxl19, 2021-11-04:
Sure you can. But you may need to estimate the translation part of the pose by IMU integration.

AdamPengG, 2021-11-04:
I got it. Thank you for your generous.

closed 2021-11-04 by AdamPengG