zxl19/Hand_Eye_Extrinsic_Calibration #3: A little confuse about the lidar coordinate system and odomtery coordinate system

AdamPengG, 2021-11-16:
Hi Zhang, thanks for your previous generous help. I have confuse about the lidar coordinate system and odomtery coordinate system. My lidar coordinate system is x(back),y(left),z(up). Is the topic, /aft_mapped_to_init, coordinate system is the same  x(back),y(left),z(up)?

zxl19, 2021-11-16:
Yes, you are right. The LiDAR coordinate system is essentially the same as odometry coordinate system. Because the points in a LiDAR scan is in the LiDAR coordinate system. And LiDAR odometry estimates the poses of LiDAR coordinate system based on the input points. However in certain algorithms (e.g. LeGO-LOAM) they are not the same because the input points are transformed before pose estimation.

AdamPengG, 2021-11-17:
You mean that ,in the lego-loam, there is a rotation between LIDAR coordinate system and LIDAR Odometry  coordinate system. If I used lego-loam to output the odometry,  the transformation matrix from matlab shows relation between LIDAR Odomety and INS . If I want to get the transformation between Lidar and INS, then I should use Matlab transformation matrix * Lego-loam Lidar&Lidar Odometry's rotation matrix. However, A-LOAM will directly output the transfromation between the Lidar and INS.

zxl19, 2021-11-17:
LeGO-LOAM transformed the point cloud during the process:

```cpp
point.x = segmentedCloud->points[i].y;
point.y = segmentedCloud->points[i].z;
point.z = segmentedCloud->points[i].x;
```

You can refer to [featureAssociation.cpp#L500](https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/blob/896a7a95a8bc510b76819d4cc48707e344bad621/LeGO-LOAM/src/featureAssociation.cpp#L500) for more details.

AdamPengG, 2021-11-23:
It works. Thank you very much ,Zhang.

closed 2021-11-23 by AdamPengG