zxl19/Hand_Eye_Extrinsic_Calibration #5: About Z-axis drift

AdamPengG, 2021-12-27:
张同学你好，
我自己使用lego-loam录制了轨迹，用来和INS进行配准。发现旋转heading,pitch,roll和XY的平移误差都比较小，XY坐标平均误差偏移在5cm以内，但是Z轴的偏移比较高，平均误差大约是25cm左右，请问你也有遇到过这样的问题吗？

zxl19, 2021-12-27:
@AdamPengG 您好！问题来源可能有两方面，一是LeGO-LOAM的位姿估计结果在Z方向上误差较大，这是激光里程计普遍存在的问题，但是在标定场景这种小范围场景下误差通常可以接受；二是由于激光雷达和INS的Z坐标轴方向基本上重合，车辆在在运动中近似平面运动，垂直方向位姿增量不明显，导致手眼标定获得的垂直方向的外参不准确，我在自己测试中获得的垂向外参通常在-1m（规定的优化变量的下限处），可以将垂直方向的外参替换为测量值，或者采用激光雷达-相机-INS联合标定获得更好的外参标定结果。

AdamPengG, 2021-12-27:
感谢！您的意思，是可以通过物理测量INS和Lidar的Z轴平移变量。然后用物理测量的Z轴偏移量，去替换程序计算出来的平移矩阵中的Z方向的偏移，旋转和XY的偏移还是依照计算出结果。

zxl19, 2021-12-27:
@AdamPengG 是的，您可以先这样尝试一下，看这组外参是否满足应用要求。

AdamPengG, 2021-12-27:
了解，感谢。还有一个问题，关于下面的代码：
`lb = [-1, -1, -1, -2, -2, -2, -2];
ub = [1, 1, 1, 2, 2, 2, -2];
x0 = (lb + ub)/2; % x y z (m) qw qx qy qz`
最后一个qz的上边界和下边界为什么都是-2呢。

zxl19, 2021-12-27:
@AdamPengG 这里应该是我的笔误，您是在哪个文件中发现的，我检查一下。

AdamPengG, 2021-12-27:
在 [Hand_Eye_Extrinsic_Calibration/main_calibration_L2I_quat_interp.m ](https://github.com/zxl19/Hand_Eye_Extrinsic_Calibration/blob/main/main_calibration_L2I_quat_interp.m)中发现的。

zxl19, 2021-12-27:
@AdamPengG `qz`的上限应该改为2，这里是我的笔误。

AdamPengG, 2021-12-27:
感谢您慷慨的解答。

zxl19, 2021-12-27:
不客气。

closed 2021-12-27 by zxl19

AdamPengG, 2021-12-30:
能方便加您个微信吗？方便以后交流。My wechat:424158674