zxl19/Hand_Eye_Extrinsic_Calibration #6: Lidar-GNSS Calibration

TheTomer, 2022-10-19:


zxl19, 2022-10-19:
Hi @TheTomer , you can refer to the [Input Pose Format](https://github.com/zxl19/Hand_Eye_Extrinsic_Calibration#2-input-pose-format) section in README.md for more details. Basically you need a LiDAR SLAM algorithm to estimate LiDAR poses and convert LiDAR and GNSS poses into required format before using this repository for calibration. If you encounter further problems, please let me know.

Best regards.

zxl19, 2022-10-19:
Hi @TheTomer , `qw qx qy qz` is a [quaternion](https://en.wikipedia.org/wiki/Quaternion) representing rotation. 

$\boldsymbol{q} = q_{w} + q_{x}\boldsymbol{i} + q_{y}\boldsymbol{j} + q_{z}\boldsymbol{k}$

closed 2022-10-29 by zxl19