zxl19/Hand_Eye_Extrinsic_Calibration #7: Question about bounds in quat/quat_interp LiDAR INS optimization

heethesh, 2024-01-11:
Did you mean to set the upper bound for qz to +2 instead of -2 here:
https://github.com/zxl19/Hand_Eye_Extrinsic_Calibration/blob/main/main_calibration_L2I_quat_interp.m#L76C1-L77C29

```
lb = [-1, -1, -1, -2, -2, -2, -2];
ub = [1, 1, 1, 2, 2, 2, -2];
```
Same for `main_calibration_L2I_quat`.

heethesh, 2024-01-11:
I'm also curious why you chose not to optimize in the so(3) space of rotations (the tangent space of rotation matrices)? That would also make interpolation simpler and avoids normalization of Quaternion in a couple of places including the state vector in loss function and results.

zxl19, 2024-01-11:
@heethesh Hi, thank you for pointing out this upper bound question. It's a typo and the upper bound for $q_z$ should be 2.

closed 2024-01-13 by heethesh