zxl19/Hand_Eye_Extrinsic_Calibration #8: lidar2rtk  calibration

pedestrain123, 2025-01-27:
您好请问基于手眼标定原理进行lidar2rtk的外参标定三轴角和平移精度可以达到多少？如果车辆较大，只能进行平面绕8字运动？还有一个问题，您有做过基于优化的方法进行两者外参标定吗？类似于apollo和商汤开源的lidar2ins那种方法，基于组合惯导位姿和初始外参把点云转换到世界坐标系下，然后进一步优化得到外参？期待您的回复

zxl19, 2025-02-06:
您好，感谢关注！

1. 外参标定问题一般没有真值可供精度评估（如果有真值那就不需要外参标定了），因此在精度评估时，通常采用多次采集数据标定，评估多次标定结果一致性的方式评估外参标定精度，此外，CAD数据也可以在一定程度上提供参考；
2. 如果车辆较大，只能做平面运动，可以在优化时固定高度方向上的外参，或者高度方向上的外参使用测量值；
3. 本仓库使用的也是基于优化的方法，但是由于开发时间较早，加上当时个人能力和调研范围比较有限，没有尝试过你提到的相关方案，可以提供一下相关的论文或仓库链接吗？

希望能够解决你的疑问，祝好。

pedestrain123, 2025-02-07:
> 您好，感谢关注！
> 
> 1. 外参标定问题一般没有真值可供精度评估（如果有真值那就不需要外参标定了），因此在精度评估时，通常采用多次采集数据标定，评估多次标定结果一致性的方式评估外参标定精度，此外，CAD数据也可以在一定程度上提供参考；
> 2. 如果车辆较大，只能做平面运动，可以在优化时固定高度方向上的外参，或者高度方向上的外参使用测量值；
> 3. 本仓库使用的也是基于优化的方法，但是由于开发时间较早，加上当时个人能力和调研范围比较有限，没有尝试过你提到的相关方案，可以提供一下相关的论文或仓库链接吗？
> 
> 希望能够解决你的疑问，祝好。

我这边手头有下载几篇论文，如果方便的话的您加我一下微信，我微信发给您，19516191665

zxl19, 2025-02-10:
已添加，本问题将关闭，您可以删除个人信息以保护隐私。

closed 2025-02-10 by zxl19